OH1KK/KiwiSDR-antenna-switch-extension #18: Update loop on kiwi version 1.95

HuubH, 2017-06-16:
Today i found that my KiwiSDR was caught in an update loop. After removing the antenna switch  extension and a rebuild of de KiWi server updating works fine. 
Then i tried to reinstall the server and it worked to some degree. The extension doesn't show up within the KiWi web environment but the antenna does switch when i use the shell command. I took a look at the demo-site and found it updating too. Not sure it is the same loop problem.

regards Huub

OH1KK, 2017-06-16:
I can reprocude problem. Something has changed in KiwiSDR. I'll look at this.

OH1KK, 2017-06-16:
Updated ant_switch.c. Should now work with v1.95.

Could you please test and confirm if it's working with your KiwiSDR ?

HuubH, 2017-06-16:
Hi,

updated (no errors seen)  a reboot, and all is working again!

thanks!!
Regards Huub

kiwisdr.hameleers.net

jks-prv, 2017-06-16:
I'm sorry guys. I didn't realize ant_switch.c used non_blocking_cmd(). In v1.95 I changed those routines to finally eliminate all the fixed static buffer use.

HuubH, 2017-06-16:
One last small issue, not a big thing :-) 
I just restarted the server from the admin page and the switch stopped working. Only  a  cold boot (power off/on) got the switch working again. So it seems like a full power off/on cycle is needed to get things running after a soft restart.

Amazing stuff you guys create by the way, Thanks!

Huub

OH1KK, 2017-06-17:
This ticket was about update loop and it is now fixed. I closethis ticket. Cold boot needed is another issue. Please open separate ticket for that.

After warm boot, did you try to reload page on a web browser with shift+f5?

When opening a new ticket please describe your hardware setup. What kind of antenna switch hardware you are using and if you use direct beagelbone gpio pins, tell what GPIO pins you are using. Hardware troubleshooting is difficult to me because my only KiwiSDR hardware is located offsite.

closed 2017-06-17 by OH1KK