OH1KK/KiwiSDR-antenna-switch-extension #19: KiwiSDR antenna switch extension init after restart failed

dl2sba, 2017-10-01:
I found one issue with this extension.

After restart of the Kiwi server (for example after FW update) the extension isn't working.
You have to reboot the beagle bone to get it working.

In the web interface the buttons can be clicked, but they stay white and no change on the GPIO pins.

Running v1.128

dl2sba, 2017-10-02:
With latest v1.132 still the same problem.

OH1KK, 2017-10-28:
GPIO things are defined in /root/extensions/ant_switch/backends/ant-switch-backend-beagle-gpio file. There are line

SAFEGPIOS=(45 44 23 26 47 46 27 65 22 61)

Which defines what GPIO's are used by atenna switch.  

# P8 pin 11 - GPIO_45 - Antenna 1
# P8 pin 12 - GPIO_44 - Antenna 2
# P8 pin 13 - GPIO_23 - Antenna 3
# P8 pin 14 - GPIO_26 - Antenna 4
# P8 pin 15 - GPIO_47 - Antenna 5
# P8 pin 16 - GPIO_46 - Antenna 6
# P8 pin 17 - GPIO_27 - Antenna 7
# P8 pin 18 - GPIO_65 - Antenna 8
# P8 pin 19 - GPIO_22 - Antenna 9
# P8 pin 26 - GPIO_61 - Antenna 10

Some of those GPIO's most likely conflict with KiwiSDR on boot time. Unfortunately I do not have spare KiwiSDR hardware to debug this problem. My only KiwiSDR is "on production" and located offsite.

There a are discussions about safe GPIO's on http://valentfx.com/vanilla/discussion/590/kiwisdr-antenna-switch-extension-and-beaglebone-black-green-gpios

If you hardware on our hands, you can try modify  /root/extensions/ant_switch/backends/ant-switch-backend-beagle-gpio file and change used GPIO pins 

SAFEGPIOS=(45 44 23 26 47 46 27 65 22 61)

GPIO's 39 - 46 might have confict with boot, so you could try for example

SAFEGPIOS=(23 26 47 46 27 65 22 61)

dl2sba, 2017-10-28:
Tnx Kari for feedback

I think the reboot is not the problem, because all works fine when I completely reboot the device.

The init fails, when only the Kiwi-SW is restarted - i.e. in case of an firmware upgrade. 

This looks for me more as an init problem of the extension - your thoughts?

OH1KK, 2017-10-28:
Maybe KiwiSDR changes GPIO settings (direction out->in) on update. I don't know. I tried to change AntSW_Initialize() function. Could you try this one?

https://raw.githubusercontent.com/OH1KK/KiwiSDR-antenna-switch-extension/master/backends/ant-switch-backend-beagle-gpio

dl2sba, 2017-10-28:
Now I get this in the log

Sat Oct 28 11:35:53 2017 0:01:33 0...      ### ant_switch_msgs RX0 <SET ext_server_init>
Sat Oct 28 11:35:53 2017 0:01:34 0...      ### ant_switch_msgs RX0 <GET Antenna>
Sat Oct 28 11:35:53 2017 0:01:34 0...      ant_switch_queryantenna BAD STATUS? <?0>
Sat Oct 28 11:35:56 2017 0:01:37 0...      ### ant_switch_msgs RX0 <SET Antenna=1>
Sat Oct 28 11:35:56 2017 0:01:37 0...      ### ant_switch_msgs RX0 <GET Antenna>
Sat Oct 28 11:35:56 2017 0:01:37 0...      ant_switch_queryantenna BAD STATUS? <?0>

and now its even not working after complete system reboot ...

OH1KK, 2017-10-28:
Starange. I cannot reproduce problem. Does antenna swtiching work from ssh console? Do you see errors?

/root/extensions/ant_switch/frontend/ant-switch-frontend s
/root/extensions/ant_switch/frontend/ant-switch-frontend 1

dl2sba, 2017-10-28:
Shame on me - I just forgot chmod 755: - /

Now it works, but I want to watch it during the next days.

closed 2017-10-28 by dl2sba