OH1KK/KiwiSDR-antenna-switch-extension #29: Some minor issues with the antenna switch extension

SM2BYC, 2019-02-07:
I have noticed some minor issues with the antenna switch extension:

1. With 8 antennas defined, the description for the 8th antenna is located partly outside the extension area. Perhaps the layout could be changed slightly to accommodate also antenna 8 within the extension area.
2. When all four Rx are in use and the last user activates the antenna switch extension, the extension shows selected antenna: unknown and permissions: unknown. Also the current antenna is not shown. I think it is still possible to select antennas but no feedback is given.

[image]

3. As the antenna switch extension is very popular, the log is often filled with "GET Antenna" entries. Would it be possible to suppress these messages from the log? The "SET Antenna" entries are however of interest as they show which antennas are most popular.

[image]

Regards

Mauritz / SM2BYC

jks-prv, 2019-02-07:
Hi Mauritz. What browser are you using? I was the last one to fiddle with the button layout and I tuned the enclosing box height to fit (see below from Firefox). Does your extension box not have the margin and rounded corners? I notice there is no margin at the top for the close (X) and help buttons.

I can fix the stray messages in the log. The other problem I'm not sure about.

[image]

SM2BYC, 2019-02-07:
Hi John,

Using latest Firefox on a Windows 7 machine. Yes it has rounded corners but Antenna 8 description is partially outside the box. Good to hear that you can fix the log issue.

/ Mauritz

[image]

jks-prv, 2019-02-07:
Okay. I'm looking at the code now and I see that I actually compute the "optimal" height with a formula to enclose the configured number of buttons. So it's obviously not doing the right thing on Windows. I have a W10 machine here so I can check it on that.

SM2BYC, 2019-02-07:
Checked on of my Win10 machines and it has the same issue with Antenna 8.

jks-prv, 2019-02-08:
My W10 machine was fine, but this might due to differences with browser fonts.

Anyway, all these problems have been fixed in my fork and a pull request has been submitted to Kari.

SM2BYC, 2019-02-08:
OK, Thanks John.

closed 2019-03-18 by OH1KK