OH1KK/KiwiSDR-antenna-switch-extension #32: Switching antenna from http url with a parameter in mixed mode

OH1KK, 2019-03-18:
Current implementation on antenna switching on url parameter allow you to select antenna. 

http://sdr.vy.fi/?ext=ant,3

This works fine and is logical if you use just one antenna. If you are using antenna switch extensions in mixing mode where selecting multiple antenna simultaneously is allowed, this does antenna toggling instead of selecting single antenna.

Should there are +3, -3 commands for toggling and 3 just to select single antenna 3. If so, how multiple selection should be handled (something like ext=ant,1,3,7).  Should there be mechanism to restrict antenna switching from url parameter. 

Leaving this open until got idea what is way to go.

jks-prv, 2019-03-19:
Good idea. I had not considered this situation. I was just trying to respond to a simple use case as requested from the Kiwi forum: http://valentfx.com/vanilla/discussion/1511/is-it-possible-to-pass-keyboard-shortcuts-via-a-url-fixed-in-v1-274

jks-prv, 2019-03-23:
Fixed in my latest pull request.